respeaker/get_started_with_respeaker #173: Respeaker (array mic hat, V2, rasp 4mics)

PierreAlbertPro, 2017-10-10:
Hi.

I have a few questions.
There is a version 2 of the core module on the github but none on the seeed.cc shop. 
Is it available to buy? If so where?

Has the onboard microphone array of the core v2 similar specs as the mic array hat ? (i.e. is it able to do far field speech recognition together with detection of the direction of speech?)

Additionally, how capable is the 4mic array raspberry hat compared to the array hat & core v2?

I plan to use a microphone array for a built-in recording setup of two persons in a room, and use the DOA for speaker identification. Which system would be suited for my use to record both audio and DOA?

So far, I see these items that could fit my needs:
core v1: https://www.seeedstudio.com/ReSpeaker-Core-Based-On-MT7688-and-OpenWRT-p-2716.html
core v2: ?
Microphone array hat: https://www.seeedstudio.com/ReSpeaker-Mic-Array---Far-field-w%2F-7-PDM-Microphones--p-2719.html
raspberry hat: https://www.seeedstudio.com/ReSpeaker-4-Mic-Array-for-Raspberry-Pi-p-2941.html)

Kind regards
Pierre Albert

Pillar1989, 2017-10-11:
@yexiaobo-seeedstudio

yexiaobo-seeedstudio, 2017-10-13:
@PierreAlbertPro Hi, 

>>There is a version 2 of the core module on the github but none on the seeed.cc shop.
Is it available to buy? If so where?

V2 still under development. We just provide some samples for alpha users. It will available end of this Oct or mid-Nov.

>>Has the onboard microphone array of the core v2 similar specs as the mic array hat ? (i.e. is it able to do far field speech recognition together with detection of the direction of speech?)

There are 6-mic on V2 with two AC108 ADC which be used in the ReSpeaker 4-Mic Array Pi HAT.
It's the powerful ReSpeaker development board ever, of course it can do far field speech recognition and other voice algorithms, such DoA, BF and AEC.

>>Additionally, how capable is the 4mic array raspberry hat compared to the array hat & core v2?

ReSpeaker 4-Mic Array is designed for Raspberry Pi 2 or later version. And V2 is All-in-one voice development platform which based on the [Rockchip's RK3229](http://www.rock-chips.com/a/en/products/RK32_Series/2016/1109/799.html), a quad-core 1.5Ghz high performance SoC.

Besides, V2 is full open source, not only hardware, but also OS, drivers and voice algorithms.

>>I plan to use a microphone array for a built-in recording setup of two persons in a room, and use the DOA for speaker identification. Which system would be suited for my use to record both audio and DOA?

1. ReSpeaker MicArray with ReSpeaker Core/PC/Mac/Linux is Ok for you requirement.
1. ReSpeaker 4-Mic Array + Raspberry Pi 3.
1. ReSpeaker V2, it not ready yet, so maybe you have to wait a few weeks.

PierreAlbertPro, 2017-10-18:
Thank you for the detailed reply. We've ordered the ReSpeaker MicArray to give it a try.
We'll see when we need to order more if we go for the v2 or the MicArray.

closed 2017-10-18 by PierreAlbertPro