respeaker/get_started_with_respeaker #178: Mopidy missing from install

devanl, 2017-10-21:
I took the respeaker out of box and followed the upgrade per 
http://wiki.seeed.cc/ReSpeaker_Drive_Unit/
and then the setup per 
http://wiki.seeed.cc/ReSpeaker_Core/

I noticed early on that the dashboard wasn't really working per the instructions.  The console, wireless and update are present but no other options are available.  

```
root@ReSpeaker:~# /etc/init.d/mopidy stop
-ash: /etc/init.d/mopidy: not found
root@ReSpeaker:~# /etc/init.d/alexa stop
-ash: /etc/init.d/alexa: not found
root@ReSpeaker:~# python playmusic.py
Traceback (most recent call last):
  File "playmusic.py", line 5, in <module>
    from respeaker import Microphone
  File "/usr/lib/python2.7/site-packages/respeaker-0.6.0-py2.7.egg/respeaker/__init__.py", line 18, in <module>
    from respeaker.microphone import Microphone
  File "/usr/lib/python2.7/site-packages/respeaker-0.6.0-py2.7.egg/respeaker/microphone.py", line 33, in <module>
    import pyaudio
ImportError: No module named pyaudio
```

Is there a guide to installing these?

Thanks,
Devan

Pillar1989, 2017-10-23:
To reduce the image size, some packages has been removed. There's also a tarball file named package.tar.bz2 which contains the removed packages. We can download it and install the packages manually if we really need them.

prehensile, 2017-10-23:
This is essentially the same issue as the one I reported in #177 - I've suggested a pull request to make the docs a bit clearer.

devanl, 2017-10-24:
I have the package tarball extracted, how do I get mopidy set up like the quickstart pictures show?

Pillar1989, 2017-10-26:
No, webui have been installed , 
We didnt show 'music','files' default. Please see
https://github.com/respeaker/frontend/commit/0ecb686e0d93f74b491ee976cab4afc20139f0dd

closed 2017-11-13 by KillingJacky