respeaker/get_started_with_respeaker #180: After update of image, incorrect version reported

devanl, 2017-10-26:
I installed the image (`ramips-openwrt-v1.0.01-LinkIt7688-squashfs-sysupgrade.bin`) to support drive unit but the login banner seems to indicate that the version does not match.

> BusyBox v1.23.2 (2017-04-18 10:22:05 CST) built-in shell (ash)
> 
>            _       __
>           |_)  _  (_  ._   _   _. |   _  ._
>           | \ (/_ __) |_) (/_ (_| |< (/_ |
>                       |
>  -----------------------------------------------------
>  CHAOS CALMER (Chaos Calmer, v0.9.99)
>  -----------------------------------------------------
>   * 1 1/2 oz Gin            Shake with a glassful
>   * 1/4 oz Triple Sec       of broken ice and pour
>   * 3/4 oz Lime Juice       unstrained into a goblet.
>   * 1 1/2 oz Orange Juice
>   * 1 tsp. Grenadine Syrup
>  -----------------------------------------------------
>  issue: http://www.seeed.cc/respeaker
>  -----------------------------------------------------
>  default: username:root password:root
>  -----------------------------------------------------
> root@ReSpeaker:~#
> 

Shouldn't this banner be reporting 1.0.01 if the upgrade succeeds?

KillingJacky, 2017-10-29:
Thanks for reporting, this is literally a negligence, but not a critical issue for usage, will fix this in the future release

devanl, 2017-10-29:
Is there a method to confirm that the install was successful?

Thanks,
Devan

KillingJacky, 2017-11-13:
v1.0.01 has stripped some packages to reduce the firmware image size, e.g. mopidy, run command `which mopidy`, if nothing found, you're on the new firmware then.

closed 2017-12-15 by KillingJacky