respeaker/get_started_with_respeaker #181: Noise issues with ReSpeaker Far-field Mic Array

PierreAlbertPro, 2017-10-27:
Hi

I've just received the microphone array module and I am having some audio issues (directly connected to a windows 10 computer. The same is observed connecting the array on a device with Ubuntu).
When listening/recording with Audacity/recording using the provide example with pyaudio, white noise starts to appear.
This is the behaviour I observe:
- if speaking in back(=where the usb cable is attached) or front of the device, the audio is clear.
- if speaking on either side, white noise appears

The duration of the noise is often silmilar, a few seconds (5-8), then dies out and white noise starts again. 

I tried turning off agc.

I ran a mic test using a 1khz tone generator
first try> node michid.js get 0x02 2
2 , 80 , 2 , 0 , 1 , 0 , 0 , 0
second and further tries after rebooting the array> node michid.js get 0x02 2
2 , 80 , 2 , 0 , 1 , 7d , 0 , 0

The firmware is the #31, but I did not try to update to the #32(20170329) nor #32(20170324) since I'm nore sure where to get the #31 version if I have to rollback.

For the  same reason, I haven't tried yet to check individual microphones using the raw #83 firmware.

I've put sample recordings here : https://soundcloud.com/user-239230166/sets/respeaker-mic-array-noise-tests

xiongyihui, 2017-11-20:
You should try the raw firmware https://github.com/respeaker/mic_array_dfu
Don't worry. You wouldn't break it.

gdpeter8075, 2017-12-14:
Do you understand what those number means?
for example: 2 , 80 , 2 , 0 , 1 , 0 , 0 , 0
I know the first 2 should be reg address

KillingJacky, 2018-01-29:
Will close this issue, please open a new issue if you have further questions.

closed 2018-01-29 by KillingJacky

PierreAlbertPro, 2018-01-30:
Just a quick comment for anyone with the same problem. 
We returned the microphone, and got a functional unit after three exchanges (3 faulty mics ko, 3 faulty mics, 4 faulty mics). Our guess is that our supplier had a defective batch since the last (working) unit comes from a resupply.

Devanshi1510, 2019-08-02:
Hi @xiongyihui  Will this https://github.com/respeaker/mic_array_dfu   firmware work for respeaker mic array v2.0 board?