respeaker/get_started_with_respeaker #183: fsck "not found" and not executable

roberteshelton, 2017-11-06:
I received several messages during boots that one of the file systems was corrupt and "run fsck". I'm accustomed to being able to execute fsck from a root shell. Typed fsck and received a "not found" message. Did find on fsck and found 2, then later 3, copies: /lib/functions/, /overlay/upper/lib/functions and /rom/lib/functions. The first seemed most likely to be executable; but had rw/r/r permissions. None of those were in the $PATH. Tried chmod on the file, but permissions didn't change. Tried executing as a shell script with sh, but nothing happened. What's up with the OS install?

roberteshelton, 2017-11-06:
Peripheral observation: the software seems very fragmented and unstable. I've attempted 192.x.x.x/home.html, but have to fiddle to get the WiFi to connect. Then Next goes to a page that says sign up at seed (to do); and next from there goes to a page that says Connect ReSpeaker to your drive unit via Aux. Well, my ReSpeaker plugs into the drive unit, and the MK drive unit has no aux port. And there is nothing else on that screen. Dead end.

Sometimes the unit cycles the USB, causing my Win10 to ping repeatedly as the serial port goes up and down.

Hardware does not seem stable either. I'm back to ground zero bc of multiple previous crashes that left the unit in some odd state not connecting to the WiFi. I could not figure out how to reconnect using Juci Box. I've worked on many Linux systems and with openwrt on multiple routers. ReSp seems very brittle in comparison.

Pillar1989, 2017-11-14:
@roberteshelton  ReSpeaker v1 installed OpenWrt os.

Pillar1989, 2017-11-14:
About how to use the board .Please refer to https://github.com/respeaker/get_started_with_respeaker/blob/master/docs/ReSpeaker/README.md

closed 2017-12-15 by KillingJacky