respeaker/get_started_with_respeaker #184: Record in mono (1 channel)

piLinux, 2017-11-09:
**I need to record in mono.**

`arecord -D plughw:1,0 -f S16_LE -c1 -r 16000 -v mono test1.wav`
It does not show any error but it creates a file without any audio (only noise).

I tested the same command for stereo and it worked perfectly.
`arecord -D plughw:1,0 -f S16_LE -c2 -r 16000 -v mono test1.wav`

Any idea how I can record in mono?

Thank you.

KillingJacky, 2017-11-13:
Where do you plug the mic array to? ReSpeaker Core or PC?

piLinux, 2017-11-13:
Ok here is the latest update.

I first tested with Raspberry Pi 3B (Jessie lite OS). At S16_LE (1 channel) there was no voice but only noise.
I tried with Stretch image and same result.

Later,
I tried with PC (ubuntu 16.04) and it worked. But there were noise at all configurations:

- S24_3LE (2 channels)
- S16_LE (2 channels)
- S16_3LE (1 channel) -> with more than enough noise

Only on windows with audacity at default config [ S24_3LE (2 channels) ], the recording was clear.

xyu6, 2017-11-24:
Hi Dear Customer, i first use 
[ get device script](https://github.com/SeeedDocument/ReSpeaker_Mic_Array/raw/master/res/Respeaker_Mic_array_Scripts/2_Channel_XVSM_firmware_Python_Scripts/Record_Audio/XVSM_RecordAudio_Ubuntu_Raspberry_Mac_getDeviceInfo.py) to get the respeaker mic array index. then i modify the index and change channel to 1 in  [record script.](https://github.com/SeeedDocument/ReSpeaker_Mic_Array/raw/master/res/Respeaker_Mic_array_Scripts/2_Channel_XVSM_firmware_Python_Scripts/Record_Audio/XVSM_RecordAudio_Ubuntu_Raspberry_Mac_recordaudio.py) I got the mono wav and i use the audacity to play and it works well. Bill

piLinux, 2017-11-27:
Thank you a lot xyu6. It solved the issue 👍

closed 2017-11-27 by piLinux