respeaker/get_started_with_respeaker #189: WINDOWS 10 IOT CORE SUPPORT RUNNING ON RASPBERRY PIE 3

abhijitdalavi, 2017-11-30:
I want order the respeaker will it work with raspberry pie 3 running windows 10 iot core..
I have tried to sucessfully install the respeaker windows driver using powershell on windows 10 iot core reaspberry pie.
i have attached screenshot below ....what you thing will it work on windows 10 iot core??
[image]

jdzxd, 2018-01-30:
We tried, but the driver installation failed. We recommend that you use it on the Raspberry Debian system and do not need to install the driver.

jaburges, 2018-02-01:
it would be great to have a windows 10 IoT core driver - switching to Raspbian is not possible
please republish the drivers as universal drivers (shouldnt be that hard to complete)

Badger1989, 2018-08-09:
I have just been trying to get the ReSpeaker V2 working in a PI 3b+. I am able to control both the speaker and mic voloume level from the dasboard. But when I record an audio clip I just get static. I can play audio just fine. 

Running the same code on my PC I get a correct recording. 

Would be great to hear if anyone has manged to record any audio using one of these  on a PI.