respeaker/get_started_with_respeaker #192: Recording Each Microphone Individually (7 Channel wav file output)

dnakamura464, 2017-12-01:
### ReSpeaker Mic Array w/ ReSpeaker Core (not v2)

I want to be able to use the Mic Array and Core to record each microphone in the array individually so that the output wav file has 7 channels, each with a recording from a different microphone. Is this possible? Using Audacity, alsamixer, and get_device_info_by_index() in python to see how many channels I can record with, the maximum amount is always 2. 

I tried using arecord -v -D plughw:1 -f S16_LE -r 16000 -c 8 audio-8ch-16k.wav to force the mic array to record in 8 channels, but this only produced an 8-channel wav file with 2 channels having audio data, and the other 6 completely empty. 

The end goal for my application of the microphone array is to be able to calculate the position (angle and distance from the array) of people nearby by relative time delays between microphone recordings. Is there an alternate method I can use to find these same parameters?

Thanks in advance.

xiongyihui, 2017-12-01:
To get 7 channel raw audio, use the dfu tool to change the firmware of the mic array. See https://github.com/respeaker/mic_array_dfu

dnakamura464, 2017-12-01:
I stumbled upon that earlier and it worked perfectly. Thanks!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>