respeaker/get_started_with_respeaker #193: Raspberry pi zero w card doesn’t show up

nielssteenvoorde, 2017-12-06:
I did install a respeaker 2mic on me Raspberry pi zero w and did the install routine on: https://github.com/respeaker/seeed-voicecard. After install i executie the command: aplay -l and there is no seeed-voicecard listen. What could be the problem? Anyone any Idea? Could there be something wrong with soldering the male header. Bad contact? Is ther a was to fine that out. I did a clean install on Noobs wit al the fixes: sudo apt-get update, sudo apt-get upgrade first. Could the respeaker be broken?

Pillar1989, 2017-12-08:
seeed-voicecard only works on [RASPBIAN STRETCH](https://www.raspberrypi.org/downloads/raspbian/)

nielssteenvoorde, 2017-12-08:
Thanks. I didn’t now that. The new noobs 2.4.5 does have rasbian stretch in it. I’m first gonna try that one. Once again many thanks.

closed 2018-01-29 by KillingJacky