respeaker/get_started_with_respeaker #194: Meow King Drive Unit

crazyquark, 2017-12-13:
Hello,

What happened to the Meow King Drive Unit?
It looked really cool, is there any DIY alternative for making a speaker like that?

xyu6, 2017-12-22:
Hi Dear Customer, it includes a pcb board(route audio signal from respeaker core to speaker and battery management system for the inside battery), battery, speaker. thanks. Bill

closed 2017-12-22 by xyu6

ubergeek77, 2020-02-05:
@xyu6 Sorry, I hate to dig up a 2-year-old issue, but you never answered the question. *What happened to it?* The Drive Unit seems to have disappeared entirely.

Where can I buy one?

I really need one as it is the only DIY enclosure that suits my needs. It's the only thing of its kind that has ever existed.