respeaker/get_started_with_respeaker #201: ReSpeaker Mic Array - Far-field w/ 7 PDM Microphones

KaeYan, 2018-02-21:
Hi Respeaker Team, 

I just want to know if the ReSpeaker Mic Array - Far-field w/ 7 PDM Microphones has been discontinued?

xiongyihui, 2018-02-26:
The microcontroller on the mic array has been discontinued by the IC vendor XMOS.
But there is a new mic array to replace it, see https://github.com/respeaker/usb_4_mic_array
It will come soon.

scher200, 2018-03-31:
this thing looks really good [https://github.com/respeaker/usb_4_mic_array](https://github.com/respeaker/usb_4_mic_array) can we preorder it somewhere?

xiongyihui, 2018-04-11:
@scher200 The R&D of the USB 4 mic array is done, it need take some time to manufacture. Mic Array is hard to manufacture! Maybe it will be available next month. I will let you know.

scher200, 2018-04-11:
thank you for the update! Just wondering will it still fit in the same old case
[https://statics3.seeedstudio.com/seeed/img/2017-01/iSoNFiqHl9KhahvVVuGr6tsI.jpg](https://statics3.seeedstudio.com/seeed/img/2017-01/iSoNFiqHl9KhahvVVuGr6tsI.jpg)

xiongyihui, 2018-04-12:
Yes, it can fit in the case. By the way, the case design file is at https://github.com/respeaker/get_started_with_respeaker/blob/master/files/respeaker_pro_case_3d.zip
You can customize one and 3d-print it.

scher200, 2018-04-12:
@xiongyihui thank you

KillingJacky, 2018-05-25:
Will close this issue, please open a new issue if you have further questions.

closed 2018-05-25 by KillingJacky

xcraftster, 2020-02-22:
Guys, does this 7 PDM mic array works on its own? I mean, it is not necessarily to attach Respeaker Core to it, right?