respeaker/get_started_with_respeaker #202: ReSpeaker v1 on raspberry via GPIO instead of USB

Sispheor, 2018-02-24:
Hi, sorry if this question has been answered. I didn't find anything on the wiki.
I use the mic on my raspbian with USB and it works perfectly.
I would like to use GPIO of the pi to the GPIO of the mic .
Is that possible?
What is the schema then?
System settings to use in this case.

Thanks a lot!

Sispheor, 2018-02-24:
Or USB( from the pi) to GPIO on the respeaker.

xiongyihui, 2018-02-28:
The location of USB DP pin and USB DM pin on ReSpeaker Core 7688 and ReSpeaker USB 6+1 Mic Array is the same. It's at bottom-left side.

[image]

@@@@@@@@@@:point_up: 

[image]

vallettea, 2018-03-01:
Sorry if it is a dummy question but what pin can I use to connect the respeaker to raspberry pi using SPI ?

Sispheor, 2018-03-01:
So, following the USB cable pinout doc

| Pin | Name | Cable color | Description |
|-----|------|-------------|-------------|
| 1   | VCC  | Red         | +5   VDC    |
| 2   | D-   | White       | Data   -    |
| 3   | D+   | Green       | Data   +    |
| 4   | GND  | Black       | Ground      |

I need to cable 
- RED --> VDD_5V
- Black --> GND
- White --> USBDP
- Green --> USBDM

Right?

vallettea, 2018-03-01:
Sorry to interfere my question is concerning something else. Opening another issue

xiongyihui, 2018-03-02:
@Sispheor USBDP is USB Data Plus (Data+), USBDM is Data-

So, White --> USBDM, Green --> USBDM
If it doesn't work, just switch them.

Sispheor, 2018-03-02:
Thx !!!
I will try this and give you a feedback.
You should write a little doc doc about this kind of stuff I think.