respeaker/get_started_with_respeaker #203: Respeaker Core v1 not recognized by raspberry pi

vallettea, 2018-02-28:
I connected it via usb it is recognized as an arduino:
```
[ 4227.275030] usb 1-1.4: new full-speed USB device number 6 using dwc_otg
[ 4227.422326] usb 1-1.4: New USB device found, idVendor=2341, idProduct=8036
[ 4227.422340] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4227.422348] usb 1-1.4: Product: Arduino Leonardo
[ 4227.422356] usb 1-1.4: Manufacturer: Arduino LLC
[ 4227.423685] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```

I installed the driver `seeed-voicecard + ./install.sh` from the master branch and set default mic as jack.

But there is nothing when i look for an audio card:

```
root@alex:~/seeed-voicecard# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

and there is no recording device neither:
```
root@alex:~/seeed-voicecard# arecord -l
**** List of CAPTURE Hardware Devices ****
```

Any help appreciated.

xiongyihui, 2018-03-01:
@vallettea The ReSpeaker Core v1 is a standalone board which is not a sound card. The seeed-voicecard is for ReSpeaker 2 Mic Hat and ReSpeaker 4 Mic Array

xiongyihui, 2018-03-01:
Maybe transfer the audio through a WiFi network between the respeaker core and the raspberry pi.

closed 2018-03-01 by vallettea