respeaker/get_started_with_respeaker #206: How to control the LED?

reality123, 2018-03-07:
### Which version is your board? 

Note: It might be ReSpeaker (v1) or ReSpeaker Core v2, please see [the pictures](https://github.com/respeaker/get_started_with_respeaker/blob/master/README.md) to know what they look like.

### If your problematic subject is the `ReSpeaker Mic Array - Far-field w/ 7 PDM Microphones`, where do you connect it to? The ReSpeaker Core or PC?

xiongyihui, 2018-03-07:
You can try the script https://github.com/respeaker/respeaker_python_library/blob/master/respeaker/pixel_ring.py

reality123, 2018-03-07:
Thanks,now I'm trying voice recognition. only  one way that record first and then translate to display the text.do you have any methods ?