respeaker/get_started_with_respeaker #210: NodeJS SDK or API

henryoliver, 2018-04-04:
### Which version is your board? 
ReSpeaker Core V2

Hi guys! Great job on the new ReSpeaker, so excited to get my hands on it! :)

How could I listen from the ReSpeaker input mic using Node.js and communicate with a JavaScript-based Dialogflow/Google assistant SDK?

Appreciate any help on that.

Thanks!

KillingJacky, 2018-05-25:
Hi @henryoliver It's quite a common question. We now are not supporting the NodeJS SDK. If you're really a Nodejs fan, you'd better get your hands dirty to dig some useful Nodejs libraries out. Basically, you need to find a library of access the audio framework of Linux, and a library of the ASR engine.
Here're what I get from google:
- https://www.npmjs.com/package/alsa
- https://www.npmjs.com/package/pulseaudio

henryoliver, 2018-05-25:
@KillingJacky Thanks! Appreciate the help! :)

closed 2018-05-25 by henryoliver