respeaker/get_started_with_respeaker #215: How to use ODAS on mic array v1 with computer?

chad1023, 2018-05-13:
### Which version is your board? 

ReSpeaker (v1)

### If your problematic subject is the `ReSpeaker Mic Array - Far-field w/ 7 PDM Microphones`, where do you connect it to? The ReSpeaker Core or PC?

PC

### Question
I connect the microphone array with computer and run ODAS_WEB in Ubuntu, but it didn't work.
[image]

Thanks for help

xiongyihui, 2018-05-14:
Have you change the mic array's default firmware? if not, please change to the 8 channel raw audio firmware first. See https://github.com/respeaker/mic_array#requirements

chad1023, 2018-05-14:
Yes, I have changed. I run the sample code in mic_array successfully.
If I should change the config file or core file in ODAS?

xiongyihui, 2018-05-14:
Yes, you should use https://github.com/introlab/odas/blob/master/config/respeaker.cfg as the config file

chad1023, 2018-05-14:
I use the respeaker.cfg as the config file and Change the sound card number according to this:
https://github.com/respeaker/mic_array#requirements

xiongyihui, 2018-05-14:
Oh, I think we should change the cfg file a little bits.

Settings of `ssl`:
```
        # format = "undefined";
        format = "json";

        interface: {
            # type = "blackhole";
            type = "socket"; ip = "127.0.0.1"; port = 9001;
        };
```

-------------------

settings of `sst`
```
    # Output to export tracked sources
    tracked: {

        format = "json";

        interface: {
            # type = "file";
            # path = "tracks.txt";
            type = "socket"; ip = "127.0.0.1"; port = 9000;
        };

    };
```

chad1023, 2018-05-14:
I run it successfully, but there is no active source location.

[image]

xiongyihui, 2018-05-14:
The two lines are the detected sources. It seems that the sphere is not displayed in your ODAS Studio.

closed 2018-05-23 by xiongyihui