respeaker/get_started_with_respeaker #217: ReSpeaker (v1) Can't connect to Alexa Auth (http://192.168.100.1:3000)

stavrosdidakis, 2018-05-25:
ReSpeaker (v1) 

I am not able to authorise Alexa on my ReSpeaker. When I click the link, I get an Error loading page.
Can you provide some info on why this is happening and how to resolve?

In addition, is there a guide in using my AVS developer credentials on the alexa code?

Thanks

xiongyihui, 2018-05-26:
Have you tried [the avs library](https://github.com/respeaker/avs)? You can run it on your computer (Linux or Mac) first, and then run it on respeaker v1.

The README.md of the avs library has a note about changing your own credentials.

stavrosdidakis, 2018-05-26:
I don't want to spend an additional time trying to run the library on Linux or Mac, although actually I believe that the project I am working on would be much better on a RPi rather than ReSpeaker. 
In addition to the problem above, I am always getting Sector errors and can't make and edit new files, although I have additional space in SD card.

It would be therefore great if you could let me know WHY I am receiving an error loading page during the authorisation process. There is no mention in any documentation from your company as to why this occurs