respeaker/get_started_with_respeaker #218: Problem with Pyaudio

stavrosdidakis, 2018-05-25:
(ReSpeaker v1)

No matter how many examples and settings I have tried to change, I was never able to get pyaudio to work. I can't connect the interface to pyadio.

Are there any examples on how to do this properly?

Its been such a struggle to get Respeaker to do anything really - well, the lights work at least!

xiongyihui, 2018-05-26:
Could you check if `mopidy` or `alexa` is running in background?
Are you able to run `arecord`, for example, `arecord -v -f cd /tmp.s.wav`

stavrosdidakis, 2018-05-26:
mopidy and alexa are stopped.
I am able to run arecord successfully, but not pyaudio. I need by pyaudio to run the speech recognition examples - so far no luck

stavrosdidakis, 2018-05-27:
Here is my log when I try to see what's going on with pyaudio:

available devices:
(0, 'MTK APSoC I2S: - (hw:0,0)', 0L, 2L)
(1, 'sysdefault', 0L, 128L)
(2, 'default', 0L, 128L)
(3, 'dmix', 0L, 2L)

default input & output device:
Traceback (most recent call last):
  File "audioTest.py", line 14, in <module>
    print(pyaudio_instance.get_default_input_device_info())
  File "/usr/lib/python2.7/site-packages/pyaudio.py", line 949, in get_default_input_device_info
    device_index = pa.get_default_input_device()
IOError: No Default Input Device Available

stavrosdidakis, 2018-05-27:
Reinstalled firmware and it works

closed 2018-05-27 by stavrosdidakis