respeaker/get_started_with_respeaker #220: Thinking Of Buying But Have Questions

mrdgmedia, 2018-05-30:
I foolishly bought a Matrix with a ESP32 chip in it.  On the crowdfunding campaign it clearly said it was a standalone.  Fast forward, to make a long story short, it isn't.  In fact, you are REQUIRED to set it up using a Raspberry Pi.  Something that the makers of the board NEVER mentioned.  Then, it is complicated and requires you know coding in order to use it.  Still, despite all that you are unable to use it with a Windows 10 machine.  Totally useless and a waste of money.

I am still needing a far-field microphone array.  My question is, does ReSpeaker work out-the-box so to speak?  Am I required to know coding or does it require a Raspberry Pi to use it?  Does it have a simple bluetooth connection whereas I can connect it to a computer?  (That was my mistake with Matrix, I assumed a integration was as simple as connecting to bluetooth)

I am seriously considering buying Respeaker however I refuse to get burned again.  I need to know these answers before making my decision.  Thanks everyone.

xiongyihui, 2018-05-31:
There are several respeaker products.
If you just want a far field USB microphone array, [the respeaker usb 4 mic array](http://wiki.seeedstudio.com/ReSpeaker_Mic_Array_v2.0/) may meet your need. It as a sound card works well with windows 10. It doesn't require coding.

If you want a standalone microphone array, [the respeaker core v2](http://wiki.seeedstudio.com/ReSpeaker_Core_v2.0/) is an option. It has Wi-Fi, bluetooth, Ethernet, USB Host and USB OTG, it is running Debian Linux. It does require some knowledge of Linux and coding.

mrdgmedia, 2018-05-31:
I appreciate the response @xiongyihui 

Is this all I need?  Or do I need to buy additional hardware for this?  I am asking cause on YouTube I saw an unboxing and it appears to have plugs that plug into something. 

[image]

xiongyihui, 2018-05-31:
Yeah, you just need a computer with USB port to use with the mic array. Its plug is for extra extension. You don't have to use it if you don't need it.

mrdgmedia, 2018-05-31:
Yes thats fine, ultimately i wanted to use it as a mic for my PC. I just wanted to make sure i didnt need additional hardware.