respeaker/get_started_with_respeaker #221: main pinouts for Respeaker 4 mic array

IshantPundir, 2018-07-08:
Hi, I am using ReSpeaker 4 Mic Array for my robotics project and I am using raspberry pi zero W with it.
now according to my design I can not just attach my Respeaker right on top of my PI, so  basically I can add a extension cable connecting my Pi's GPIO to Respeaker's Pinouts

but I am also using a TFT display which is using some of my GPIOs 

I know "ReSpeaker 4 Mic Array " uses  I2c but just to be clear and safe I want to know what are the main GPIOs I need to connect to pi without wasting much of my GPIO, 

### If your problematic subject is the `ReSpeaker Mic Array - Far-field w/ 7 PDM Microphones`, where do you connect it to? The ReSpeaker Core or PC?

xiongyihui, 2018-07-09:
The used pins the SPI for LEDs and I2S for the codec. I2C, GPIO12 and GPIO13 are connected with Grove connectors and are available for other purpose.

Your TFT display may use SPI. Fortunately, SPI can be used by multiple devices with different chip selection pin.

[image]

You can find the schematic at http://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/#resources

IshantPundir, 2018-07-09:
hi thank you for the help
since i don't need the Led array ...
is my DOA is going to work if I only use PIN number [3,5, 12,35, 38,40]

and what is the use of GPIO 12 and 13

xiongyihui, 2018-07-09:
The DOA will work if the sound card works. GPIO 12 and 13 are used for a Grove Connector (http://wiki.seeedstudio.com/Grove_System/). You can use these two if don't use the Grove connector.

IshantPundir, 2018-07-09:
hi just one last question,
by sound card you means the I2S pins right, since according to my understanding the ADC is using I2S connection.

soo I just need to connect to I2S connections only, in order to get audio data from the array

IshantPundir, 2018-07-09:
thank you for you help 

really appreciate it , have a nice day

KillingJacky, 2018-08-06:
Will close this issue, please open a new issue if you have further questions.

closed 2018-08-06 by KillingJacky

kothapallysandeep, 2019-03-14:
1)I Need 6 mic array 30 pin conn schematic. 
2) why DAC output is connecting to ADC? it is necessary?