respeaker/get_started_with_respeaker #223: manually wiring respeaker 4 mic array to raspberry pi zero w

IshantPundir, 2018-08-19:
Hi, I am using ReSpeaker 4-Mic Array , with raspberry pi zero w

and I am also using a 480x360 lcd like this [---> https://www.amazon.in/Touch-Screen-Display-Raspberry-Pi/dp/B01N0LFGUL ]

since I can not attach raspberry pi with lcd and respeaker at the same time I have decided to manually wire the respeaker and lcd to my pi using only relevant connections 

I just want to access the audio data from the 4 Chanels of respeaker so I am not using SPI (SPI is just used for led I guess)

I am only using I2S and I2c connections ie: PIN number --> [3, 5, 12, 35, 38, 40]

but on command arecord -L
I get the following output :

 ===================================================
pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
 ===================================================

can you please help me I have checked my connections the are all good and I know my respeaker is also working because I have already tested it on other pi 

can you please tell me if my connections pins selection is correct or not

and should I do ??

xiongyihui, 2018-08-20:
You can use `i2cdetect -y 1` to check if the i2c bus works or not

```
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

`i2cdetect -y 1` will detect the mic array at i2c address `0x3b`

IshantPundir, 2018-08-20:
with i2cdetect -y 1 it is not detecting my i2c 

`
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
`

according to [https://user-images.githubusercontent.com/948283/42431305-34863a62-8377-11e8-8be6-270c5d9aa180.png](url)

i2c_SDA and i2c_SCL are used on PIN number 3 and 5 and I have checked the connections and it is there and can you please tell me if there are some other pins that is also need to use for this 

thanks

xiongyihui, 2018-08-20:
Maybe it's power supply issue. Do you check the 5v and GND connections?

[image]

See also https://pinout.xyz/pinout/respeaker_4_mic_array

IshantPundir, 2018-08-21:
hi I finally managed to detect my i2c turns out that  I need to give both 5v and 3.5v to respeaker in order to make it work 

below is the output I am getting now: 
`pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- `

but now from command record -L I am getting 

`pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
playback
ac108
default`

can you please help me with this 

thanks you