respeaker/get_started_with_respeaker #226: ReSpeaker AEC issues

Ali-Amir, 2018-08-24:
### Which version is your board? 
ReSpeaker Mic Array v2

Hi I'm trying to evaluate AEC on the ReSpeaker Mic Array V2. I tried the Audacity test - playing audio sounds through speaker connected to the mic array and recording through mic array on Audacity - both with 1 channel and 6 channel firmware versions. In both of them I saw that there is some leftover sounds from the speaker.

After that I tried connecting them to an Android phone and making a Google Hangouts call and heard a clear echo throughout. I ensured that the phone picks up the speaker and the mic. That left me wondering, is the performance of AEC that bad in such an application or am I doing something wrong. Running tuning.py to query device parameters shows that all of them are set to default values (in other words AEC is enabled). I would appreciate your help with this.

Ali-Amir, 2018-08-24:
Also, in regard to that, does speaker placement and speaker itself play a role in the AEC performance?

Ali-Amir, 2018-08-25:
Thanks for your reply @xiongyihui. I have a few follow up questions:
What is a long enough time delay to see those effects?
How close / far should the speaker be from the mic? I'm guessing it's a trade-off between echo delay for AEC and echo amplitude to rest of the signal ratio.

Yjppj, 2023-11-30:
I also encountered a situation where echo cancellation was very poor