respeaker/get_started_with_respeaker #234: How to record and save an audio file using respeaker python api

WIll-Xu35, 2019-01-25:
### Which version is your board? 
ReSpeaker Mic Array v2.0

Hi all,

I'm trying to use my ReSpeaker Far-field Mic Array v2.0 to record a piece of audio. Pyaudio is the package I used to record audio file from a normal microphone, and I want to replace normal mic with microphone array and use respeaker python api instead of pyaudio. I've read the getting started page and found there is a record function, but I cannot get it to work.

Can anyone give me a simple example, how to record a piece of audio using respeaker python api (and can the record channel be 2?)?

Much appreciated!

xiongyihui, 2019-01-25:
The `record()` function don't work in the latest pypi package. It's fixed in the source code.

You can install the source code version to use the `record()` function, run the following command to try:

```
pip install -U https://github.com/respeaker/respeaker_python_library/archive/master.zip
```

WIll-Xu35, 2019-01-25:
@xiongyihui Thank you for your quick reply.

I've searched online and found this bit of code working.

`import respeaker
import wave

def task():
    mic = respeaker.Microphone()
    data = mic.listen()
    data = b''.join(data)
    record(data)

def record(data):
    f = wave.open('/home/user/wav.wav', 'wb')
    f.setframerate(8000)
    f.setsampwidth(2)
    f.setnchannels(1)
    f.writeframes(data)
    f.close()

task()`

However, I think mic.listen contains some VAD procudure. How can I record, for example, 5 seconds of audio input without any VAD.

xiongyihui, 2019-01-25:
If you install the latest library, you can use the function `record()`.

`pip install -U https://github.com/respeaker/respeaker_python_library/archive/master.zip`

```
import respeaker

mic = respeaker.Microphone()
mic.record('test.wav', seconds=10)

```

WIll-Xu35, 2019-01-28:
@xiongyihui Hi, I've installed the latest library using the given pip command (I used pip3 instead of pip). However, the record function stills seems to be broken. After I run the script, it ends immediately and the recorded file is completely empty.

WIll-Xu35, 2019-01-28:
And can I set the sampling rate to 8000? It seems that the default sampling rate is set to 16000.

xiongyihui, 2019-01-28:
As the acoustic model uses 16000 Hz sample rate, it is not a good idea to change the sample rate.

If you just want to record an audio and save it to a file, you can just use pyaudio. The following is an example from https://people.csail.mit.edu/hubert/pyaudio/

```
import pyaudio
import wave

CHUNK = 1024
FORMAT = pyaudio.paInt16
CHANNELS = 2
RATE = 8000
RECORD_SECONDS = 5
WAVE_OUTPUT_FILENAME = "output.wav"

p = pyaudio.PyAudio()

stream = p.open(format=FORMAT,
                channels=CHANNELS,
                rate=RATE,
                input=True,
                frames_per_buffer=CHUNK)

print("* recording")

frames = []

for i in range(0, int(RATE / CHUNK * RECORD_SECONDS)):
    data = stream.read(CHUNK)
    frames.append(data)

print("* done recording")

stream.stop_stream()
stream.close()
p.terminate()

wf = wave.open(WAVE_OUTPUT_FILENAME, 'wb')
wf.setnchannels(CHANNELS)
wf.setsampwidth(p.get_sample_size(FORMAT))
wf.setframerate(RATE)
wf.writeframes(b''.join(frames))
wf.close()
```

WIll-Xu35, 2019-01-29:
@xiongyihui Thank you so much for your reply.

I can do recording using the given code right now. I used pyaudio in the first place but it didn't record anything useful and I thought it was the problem with pyaudio. Now, it seems to me that there is a delay between when I plug in the mic arracy and the time it can  work properly, and this delay can be several minutes. I'm not sure if my observation is correct or it shouldn't be like this because something else is wrong.

Another problem now is that after I integrate this bit of code to my code, it seems to stuck at 'data = stream.read(CHUNK)' right after entering the for loop. I tried to call this functionality in a test script and it worked, but not in my bulky code. Do you have any idea what may have caused this?

Sorry to have so many questions. and I really appreciate your help.

xiongyihui, 2019-02-11:
When the mic array is plugged in, the OS may take some time to initialize the device. Applications like PulseAudio may also detect the device do some checks.

The reason, your code works with a webcam but doesn't work with the mic array, should be an issue of audio bufffer, the webcam's audio buffer should be large than the mic array.

WIll-Xu35, 2019-02-12:
Thank you for your answer. I'll try to solve it following your analysis.

Happy Chinese New Year! :)

closed 2019-02-12 by WIll-Xu35

sameedsajid26, 2022-07-18:
I want to record 6 channel input and have separate wave files for output from all channels; how do I modify the code for that?

Alcatraz0000, 2022-12-13:
> I want to record 6 channel input and have separate wave files for output from all channels; how do I modify the code for that?

Did you find a solution to manage that?